rxseger/rx_tools #2: Add documentation for tools

rxseger, 2016-07-16:
Would be nice to have some integrated documentation in the repository.

Most comprehensive guides I could find:

* rtl_fm: http://kmkeen.com/rtl-demod-guide/ http://manpages.ubuntu.com/manpages/trusty/man1/rtl_fm.1.html
* rtl_power: http://kmkeen.com/rtl-power/ http://manpages.ubuntu.com/manpages/trusty/man1/rtl_power.1.html
* rt_sdr: http://manpages.ubuntu.com/manpages/trusty/man1/rtl_sdr.1.html

petterreinholdtsen, 2016-10-17:
It would also be great with manual pages for the tools.  It is a soft requirement in Debian to have manual pages.

rxseger, 2016-10-18:
I'd also like man pages for rx_tools. Not sure where those .1 files in Ubuntu rtlsdr came from, can't find them anywhere in https://github.com/librtlsdr/librtlsdr (but there's lots of forks of librtlsdr).