rxseger/rx_tools #37: rx_fm not using Airspy SDR

starvald, 2016-08-28:
lee@Mate:~$ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.0-g7e5b0b5c
API Version: v0.6.0
ABI Version: v0.6-dev
Install root: /usr/local
Module found: /usr/local/lib/SoapySDR/modules0.6-dev/libairspySupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6-dev/libosmosdrSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.6-dev/librfspaceSupport.so
Loading modules... done
Available factories...airspy, null, osmosdr, rfspace, 
lee@Mate:~$ 
lee@Mate:~$ 
lee@Mate:~$ SoapySDRUtil --probe osmosdr
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device 
Using AirSpy NOS v1.0.0-rc9-0-ga56adfd 2016-06-12, samplerates: 2.5M 10M 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=airspy
  hardware=airspy

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32
  Native format: CF32 [full-scale=1]
  Antennas: RX
  Full gain range: [0, 21] dB
    LNA gain range: [0, 15] dB
    MIX gain range: [0, 15] dB
    IF gain range: [0, 15] dB
  Full freq range: [24, 1766] MHz
    RF freq range: [24, 1766] MHz
    CORR freq range:  MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float]
     * CORR - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=CORR, units=Hz, default=DEFAULT, type=float, options=(DEFAULT, IGNORE)]
  Sample rates: [2.5, 10] MHz
  Filter bandwidths: [10] MHz

lee@Mate:~$ 
lee@Mate:~$ 
lee@Mate:~$ rx_fm -M wbfm -f 99.3M | play -r 32k -t raw -e s -b 16 -c 1 -V1 -

-: (raw)

  Encoding: Signed PCM    
  Channels: 1 @ 16-bit   
Samplerate: 32000Hz      
Replaygain: off         
  Duration: unknown      

In:0.00% 00:00:00.00 [00:00:00.00] Out:0     [      |      ]        Clip:0    Found 1 device(s):
  0: airspy=0, driver=airspy, label=AirSpy AIRSPY, 

verbose_device_search((null))
Using AirSpy NOS v1.0.0-rc9-0-ga56adfd 2016-06-12, samplerates: 2.5M 10M 
Using device airspy: 
Found 1 antenna(s): RX 
Found 3 gain(s): LNA MIX IF 
Found 2 frequencies: RF CORR 
Found 2 sample rates: 2500000 10000000 
Found 1 bandwidths: 10000000 
SoapySDRDevice_setupStream failed
Failed to open rtlsdr device matching (null).
In:0.00% 00:00:00.00 [00:00:00.00] Out:0     [      |      ]        Clip:0    
Done.
------------------------------------------------------------------------------------------

CubicSDR and rtlsdr device working so i don't think it's SoapySDR

guruofquality, 2016-08-28:
@rxseger You might want to add some %s, SoapySDRDevice_lastError() to some of the error messages. My guess here is that the complex int16 format is not supported. But that error message is not getting shown.

@starvald Can you try [SoapyAirspy](https://github.com/pothosware/SoapyAirspy/wiki)? Its not 100% complete, but it should implement complex int16. The devices from the SoapyOsmo wrapper only do floats. You will have to uninstall SoapyOsmo to avoid a name conflict though.

starvald, 2016-08-28:
Thanks, using SoapyAirspy the device is now used but i have a few other problems the audio plays very slowly.

rx_fm -f 99.3M -M fm -s 170k -A fast -r 32k -l 0 -E deemp -w 2500000 -g LNA=11,MIX=5,IF=8| play -r 300k -t raw -e s -b 16 -c 1 -V1 -

using play -r 300k it's playing at the right kind of speed.
if i don't use -w 2500000 is plays the wrong station 97.4MHz rather than 99.3MHz, i think the offset tuning is a bit off.

-------------------------------------------------------

lee@Mate:~$ rx_fm -f 99.3M -M fm -s 170k -A fast -r 32k -l 0 -E deemp -w 2500000 -g LNA=11,MIX=5,IF=8| play -r 300k -t raw -e s -b 16 -c 1 -V1 -

-: (raw)

  Encoding: Signed PCM    
  Channels: 1 @ 16-bit   
Samplerate: 300000Hz     
Replaygain: off         
  Duration: unknown      

In:0.00% 00:00:00.00 [00:00:00.00] Out:0     [      |      ]        Clip:0    Found 1 device(s):
  0: device_id=0, driver=airspy, label=AIRSPY [644064dc:e0f6bcd], serial=644064dc:e0f6bcd, 

verbose_device_search((null))
Using device Airspy: device_id=0 origin=https://github.com/pothosware/SoapyAirspy 
Found 1 antenna(s): RX 
Found 3 gain(s): LNA MIX VGA 
Found 1 frequencies: RF 
Found 2 sample rates: 10000000 2500000 
Found 0 bandwidths: 
[INFO] Using format CS16.
Setting gain element LNA: 11.000000 dB
Setting gain element MIX: 5.000000 dB
Setting gain element IF: 8.000000 dB
Set bandwidth parameter 2500000 Hz.
WARNING: Failed to set offset tuning.
Tuned to 99300000 Hz.
Oversampling input by: 6x.
Oversampling output by: 1x.
Buffer size: 8.03ms
Sampling at 1020000 S/s.
Output at 170000 Hz.
In:0.00% 00:01:00.40 [00:00:00.00] Out:11.6M [    -=|=-    ]        Clip:0    ^CSignal caught, exiting!

User cancel, exiting...
In:0.00% 00:01:00.46 [00:00:00.00] Out:11.6M [     =|=     ]        Clip:0    
Aborted.
readStream read failed: -1
dongle_thread_fn terminated
lee@Mate:~$