rxseger/rx_tools #38: Does not accept SoapySDRServer device string

thehose, 2016-09-09:
I have an rtl_sdr dongle on a server and available via SoapySDRServer. It works with the gr-osmosdr utilities like osmocom_fft but not with rx-fm.

himac:~ hose$ rx_fm -d "soapy=0,driver=remote,remote=sdr" -f 98.5M
No supported devices found.
Failed to open rtlsdr device matching soapy=0,driver=remote,remote=sdr.

OR

himac:~ hose$ rx_fm -d "soapy=0,driver=remote,remote=sdr,remote:driver=rtlsdr" -f 98.5M
No supported devices found.
Failed to open rtlsdr device matching soapy=0,driver=remote,remote=sdr,remote:driver=rtlsdr.

Hose

guruofquality, 2016-09-09:
I think the issue is that ``verbose_device_search()`` attempts to find the device without any hints passed in through the device args. With remote devices, sometime the address specification is required to find the device because network discovery does not work for one reason or another.

@rxseger I could suggest to pass the argument string to SoapySDRDevice_enumerate(), however, soapySDR already performs device matching in its factory function. So my primary suggestion to delete the entire search and match operation and just replace it with ``dev = SoapySDRDevice_makeStrArgs(s)``

I tested the following diff with a remote rtlsdr which was not being discovered on its own over the multicast (possibly due to wifi for server) and it worked. Any issues with doing this instead?

```
diff --git a/src/convenience/convenience.c b/src/convenience/convenience.c
index f0bcef5..09cae27 100644
--- a/src/convenience/convenience.c
+++ b/src/convenience/convenience.c
@@ -438,49 +438,10 @@ int verbose_device_search(char *s, SoapySDRDevice **devOut, SoapySDRStream **str
        char *s2;
        char vendor[256], product[256], serial[256];
        SoapySDRDevice *dev = NULL;
-       size_t j = 0;
 
-       SoapySDRKwargs device_make_args = {0};
-       SoapySDRKwargs device_search_args = {0};
        SoapySDRKwargs stream_args = {0};
 
-       SoapySDRKwargs *results = SoapySDRDevice_enumerate(&device_search_args, &device_count);
-       if (!device_count) {
-               fprintf(stderr, "No supported devices found.\n");
-               return -1;
-       }
-       fprintf(stderr, "Found %zu device(s):\n", device_count);
-
-       for (i = 0; i < device_count; i++) {
-               //rtlsdr_get_device_usb_strings(i, vendor, product, serial);
-               //fprintf(stderr, "  %d:  %s, %s, SN: %s\n", i, vendor, product, serial);
-               fprintf(stderr, "  %zu: ", i);
-               for (j = 0; j < results[i].size; j++)
-               {
-                       fprintf(stderr, "%s=%s, ", results[i].keys[j], results[i].vals[j]);
-                       // TODO: save and match on keys?
-               }
-               fprintf(stderr, "\n");
-       }
-       fprintf(stderr, "\n");
-
-       fprintf(stderr, "verbose_device_search(%s)\n", s);
-       if (s) {
-               if (strchr(s, '=')) {
-                       // foo=bar,baz=quux key/value pairs
-                       parse_kwargs(s, &device_make_args);
-               } else if (isdigit(s[0])) {                                                                                                                                                                                                  
-                       // Passing an integer (-d 0, -d 1...), searches for nth rtlsdr for librtlsdr compatibility                                                                                                                           
-                       SoapySDRKwargs_set(&device_make_args, "driver", "rtlsdr");                                                                                                                                                           
-                       SoapySDRKwargs_set(&device_make_args, "rtl", s);                                                                                                                                                                     
-               } else {                                                                                                                                                                                                                     
-                       // Exact serial match TODO: prefix, suffix                                                                                                                                                                           
-                       // TODO: seems to not work? have rtl_eeprom programmed serial number "3" on RTL-SDR but it chooses the HackRF instead, if present                                                                                    
-                       SoapySDRKwargs_set(&device_make_args, "serial", s);                                                                                                                                                                  
-               }                                                                                                                                                                                                                            
-       }                                                                                                                                                                                                                                    
-                                                                                                                                                                                                                                            
-       dev = SoapySDRDevice_make(&device_make_args);                                                                                                                                                                                        
[... cut ...]
        if (!dev) {                                                                                                                                                                                                                          
                fprintf(stderr, "SoapySDRDevice_make failed\n");                                                                                                                                                                             
                return -1;  
```

rxseger, 2016-09-30:
👍 Looks good to me, the reason I called SoapySDRDevice_enumerate then SoapySDRDevice_make is mainly because librtlsdr worked that way, and I did a fairly direct function-by-function port from librtlsdr to SoapySDR. Only possible issues I could think of, neither too serious:

* Removes the "Found %zu device(s)" listing, which could be informative for debugging
* Removes the legacy intege match (-d 0, -d 1), preserved for compatibility with librtlsdr tool scripts

closed 2017-03-11 by guruofquality

cdgraff, 2019-03-23:
Hi @rxseger, I'm trying to use -d param to select 1 device from 4 and always see that select the same rtl=0 (first device)

When pass -d 1, 2,3 or any other combination always got an error and show this error message:

Using device UNKNOWN: origin=https://github.com/pothosware/SoapyRTLSDR rtl=0 

Some advice about what I'm doing wrong?

Thanks!

guruofquality, 2019-03-23:
The device arguments are key value pairs, so you should use the serial number to specify a specific RTLSDR:

`-d "serial=12345ABCD"`

SoapySDRUtil --find,--make,and--probe take the same optional argument string, as does gr-osmosdr blocks

cdgraff, 2019-03-23:
@guruofquality thanks for the quick response! and that works perfect! probably I misunderstood the documentation :(

commit c5ceced7 (2017-03-11): verbose_device_search passes args directly to SoapySDRDevice_makeStrArgs

SoapySDRDevice_make() ensures that an enumerated device is found
and uses the available keys and values in the hint string.
No need to implement this again in rx tools

* Resolves #38