rxseger/rx_tools #41: Incorrect description when running rx_power

petterreinholdtsen, 2016-10-17:
When running rx_power without any arguments, I see this start:

```
rtl_power, a simple FFT logger for RTL2832 based DVB-T receivers

Use:    rtl_power -f freq_range [-options] [filename]
```

This seem wrong, as the program is named rx_power and it is not only for DVB-T receivers?

rxseger, 2016-10-17:
Internally the tools are still refered to by their librtlsdr names, but I could change the descriptions in the usage messages to use the rx_tools tool names

petterreinholdtsen, 2016-10-17:
[Robert X. Seger]
> Internally the tools are still refered to by their librtlsdr names,
> but I could change the descriptions in the usage messages to use the
> rx_tools tool names

Great.  It is very confusing when the usage information talk about a
different program in a different package in Debian.

-- 
Happy hacking
Petter Reinholdtsen

closed 2016-10-18 by rxseger