rxseger/rx_tools #46: HackRF Not Found (No supported devices found. Failed to open rtlsdr device matching (null).)

scottstables, 2016-11-23:
Hi - see below.  Note that my hackrf is detected when using hackrf_info (version 2014.8.1). Any ideas ?

I get the following when issuing one of the examples.

sudo ./rx_power -f 0.1M:6G:1M -e 24h hackrf-power-1m.csv

Number of frequency hops: 5999
Dongle bandwidth: 1000000Hz
Downsampling by: 1x
Cropping by: 0.00%
Total FFT bins: 5999
Logged FFT bins: 5999
FFT bin size: 500325.12Hz
Buffer size: 16384 bytes (8.19ms)
Reporting every 10 seconds
No supported devices found.
Failed to open rtlsdr device matching (null).

guruofquality, 2016-11-23:
Quick question: Whats the output of `SoapySDRUtil --info` and is hackrf in the available factories? You might need to install SoapyHackRF

kevmiyashiro, 2017-04-03:
Hi,

I am experiencing a similar issue to OP with HackRF not being found.  Any suggestions?

Installed SoapySDR, installed SoapyHackRF.

installed rx_tools

ran soapysdrutil --info and it can see the hackrf as one of the available factories.

ran soapysdrutil --probe="driver=hackrf" and it can also see the hackrf

however when i invoke rx_power -f 88M:108M:125k out.csv I get
Number of frequency hops: 8
Dongle bandwidth: 25000000Hz
Downsampling by: 1x
Cropping by: 0.00%
Total fft bins: 256
Logged fft bins: 256
FFT bin size: 78125.00Hz
Buffer size: 16384 bytes (3.28ms)
Reporting every 10 seconds
SoapySDRDevice_make failed
Failed to open rtlsdr device matching (null).

romeojulietthotel, 2017-04-03:
Have you tried:

`./rx_power -d driver=hackrf -f 0.1M:6G:1M -e 24h hackrf-power-1m.csv`

kevmiyashiro, 2017-04-03:
sigh...it never fails to take a better read of the help options that come out from a tool.

the -d did it, thank you, sorry for the should-have-looked-around-better question!

closed 2019-03-09 by guruofquality