rxseger/rx_tools #47: rx_fm/rx_sdr - segmentation fault

jonschoning, 2017-02-18:
__1__

```
$ rx_fm -f 91500000 -s 10000000 -d driver=lime       
Found 1 device(s):
  0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 2.0] 9060B00462615, media=USB 2.0, module=STREAM, name=LimeSDR-USB, serial=0009060B00462615, 

verbose_device_search(driver=lime)
[INFO] Make connection: 'LimeSDR-USB [USB 2.0] 9060B00462615'
Estimated reference clock 30.7197 MHz
Selected reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
LMS7002M values cache at /home/jon/.limesuite/LMS7002M_cache_values.db
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Enable
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, 0 MHz)
[INFO] SoapyLMS7::setFrequency(Tx, 0, BB, 0 MHz)
[INFO] SoapyLMS7::setAntenna(Rx, 0, LNAL)
[INFO] SoapyLMS7::setAntenna(Tx, 0, BAND1)
[INFO] SoapyLMS7::setGain(Rx, 0, PGA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 0, LNA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 0, TIA, 0 dB)
[INFO] SoapyLMS7::setGain(Tx, 0, PAD, -50 dB)
[INFO] SoapyLMS7::setSampleRate(Rx, 0, 10 MHz), CGEN=80 MHz, ADC=20 MHz, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
[INFO] SoapyLMS7::setSampleRate(Tx, 0, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
[INFO] SoapyLMS7::setBandwidth(Rx, 0, 30 MHz)
[INFO] Rx Filter calibrated from cache
[INFO] SoapyLMS7::setBandwidth(Tx, 0, 30 MHz)
[INFO] Tx Filter calibrated from cache
[INFO] SoapyLMS7::setFrequency(Rx, 1, BB, 0 MHz)
[INFO] SoapyLMS7::setFrequency(Tx, 1, BB, 0 MHz)
[INFO] SoapyLMS7::setAntenna(Rx, 1, LNAL)
[INFO] SoapyLMS7::setAntenna(Tx, 1, BAND1)
[INFO] SoapyLMS7::setGain(Rx, 1, PGA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 1, LNA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 1, TIA, 0 dB)
[INFO] SoapyLMS7::setGain(Tx, 1, PAD, -50 dB)
[INFO] SoapyLMS7::setSampleRate(Rx, 1, 10 MHz), CGEN=80 MHz, ADC=20 MHz, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
[INFO] SoapyLMS7::setSampleRate(Tx, 1, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
[INFO] SoapyLMS7::setBandwidth(Rx, 1, 30 MHz)
[INFO] Rx Filter calibrated from cache
[INFO] SoapyLMS7::setBandwidth(Tx, 1, 30 MHz)
[INFO] Tx Filter calibrated from cache
Using device LimeSDR-USB: boardSerialNumber=0x9060b00462615 firmwareVersion=3 hardwareVersion=4 protocolVersion=1 
Found 6 antenna(s): NONE LNAH LNAL LNAW LB1 LB2 
Found 3 gain(s): LNA TIA PGA 
Found 2 frequencies: RF BB 
Found 2 sample rates: 100000 60000000 
Found 0 bandwidths: 
Bandwidth set to automatic resulted in 0 Hz.
[INFO] SoapyLMS7::setFrequency(Rx, 0, RF, 94 MHz)
SetFrequency using cache values vco:1, csw:173
[WARNING] GetDC_IQ_Interp(94 MHz, ch=0, tx=0): no matches between [93, 95] MHz
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, 6.10352e-07 MHz)
Tuned to 94000000 Hz.
Oversampling input by: 1x.
Oversampling output by: 1x.
Buffer size: 0.82ms
[INFO] SoapyLMS7::setSampleRate(Rx, 0, 10 MHz), CGEN=640 MHz, ADC=160 MHz, decim=16
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
Sampling at 10000000 S/s.
Output at 10000000 Hz.
[1]    9763 segmentation fault (core dumped)  rx_fm -f 91500000 -s 10000000 -d driver=lime

```

__2__

```
[1] $ rx_sdr -f 915M out                                                   ✘
Found 1 device(s):
  0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 2.0] 9060B00462615, media=USB 2.0, module=STREAM, name=LimeSDR-USB, serial=0009060B00462615, 

verbose_device_search((null))
[INFO] Make connection: 'LimeSDR-USB [USB 2.0] 9060B00462615'
Estimated reference clock 30.7197 MHz
Selected reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
LMS7002M values cache at /home/jon/.limesuite/LMS7002M_cache_values.db
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Enable
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, 0 MHz)
[INFO] SoapyLMS7::setFrequency(Tx, 0, BB, 0 MHz)
[INFO] SoapyLMS7::setAntenna(Rx, 0, LNAL)
[INFO] SoapyLMS7::setAntenna(Tx, 0, BAND1)
[INFO] SoapyLMS7::setGain(Rx, 0, PGA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 0, LNA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 0, TIA, 0 dB)
[INFO] SoapyLMS7::setGain(Tx, 0, PAD, -50 dB)
[INFO] SoapyLMS7::setSampleRate(Rx, 0, 10 MHz), CGEN=80 MHz, ADC=20 MHz, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
[INFO] SoapyLMS7::setSampleRate(Tx, 0, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
[INFO] SoapyLMS7::setBandwidth(Rx, 0, 30 MHz)
[INFO] Rx Filter calibrated from cache
[INFO] SoapyLMS7::setBandwidth(Tx, 0, 30 MHz)
[INFO] Tx Filter calibrated from cache
[INFO] SoapyLMS7::setFrequency(Rx, 1, BB, 0 MHz)
[INFO] SoapyLMS7::setFrequency(Tx, 1, BB, 0 MHz)
[INFO] SoapyLMS7::setAntenna(Rx, 1, LNAL)
[INFO] SoapyLMS7::setAntenna(Tx, 1, BAND1)
[INFO] SoapyLMS7::setGain(Rx, 1, PGA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 1, LNA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 1, TIA, 0 dB)
[INFO] SoapyLMS7::setGain(Tx, 1, PAD, -50 dB)
[INFO] SoapyLMS7::setSampleRate(Rx, 1, 10 MHz), CGEN=80 MHz, ADC=20 MHz, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
[INFO] SoapyLMS7::setSampleRate(Tx, 1, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
[INFO] SoapyLMS7::setBandwidth(Rx, 1, 30 MHz)
[INFO] Rx Filter calibrated from cache
[INFO] SoapyLMS7::setBandwidth(Tx, 1, 30 MHz)
[INFO] Tx Filter calibrated from cache
Using device LimeSDR-USB: boardSerialNumber=0x9060b00462615 firmwareVersion=3 hardwareVersion=4 protocolVersion=1 
Found 6 antenna(s): NONE LNAH LNAL LNAW LB1 LB2 
Found 3 gain(s): LNA TIA PGA 
Found 2 frequencies: RF BB 
Found 2 sample rates: 100000 60000000 
Found 0 bandwidths: 
Using output format: CU8 (input format CS16)
[INFO] SoapyLMS7::setSampleRate(Rx, 0, 2.048 MHz), CGEN=262.144 MHz, ADC=65.536 MHz, decim=32
ConnectionSTREAM::ConfigureFPGA_PLL(tx=2.048MHz, rx=2.048MHz)
Sampling at 2048000 S/s.
[INFO] SoapyLMS7::setFrequency(Rx, 0, RF, 915 MHz)
SetFrequency using cache values vco:2, csw:192
[WARNING] GetDC_IQ_Interp(915 MHz, ch=0, tx=0): no matches between [914, 916] MHz
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, 0 MHz)
Tuned to 915000000 Hz.
Reading samples in sync mode...
[1]    10299 segmentation fault (core dumped)  rx_sdr -f 915M out
```

```
$ lsb_release -a      
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.10
Release:	16.10
Codename:	yakkety

$ ldd rx_fm 
	linux-vdso.so.1 =>  (0x00007fffbe7dc000)
	libSoapySDR.so.0.6-dev => /usr/local/lib/libSoapySDR.so.0.6-dev (0x00007f4ce89e1000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4ce869f000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4ce8481000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4ce80ba000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f4ce7eb6000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4ce7b2c000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4ce7915000)
	/lib64/ld-linux-x86-64.so.2 (0x0000559f262ef000)
```

guruofquality, 2017-03-11:
I found a bug in the buffer allocation, here is the fix: https://github.com/rxseger/rx_tools/commit/886cd3a058dd7fb967513ddc978f6ad13e0f7445 Can you comment on if it fixes this issue? I fixed a similar segfault for me in rx_sdr

jonschoning, 2017-03-13:
ok, now `rx_sdr` no longer segfaults, but `rx_fm` still does

guruofquality, 2017-03-13:
This code has a lot of use of int16 arrays for complex number pairs which is the cause of some of the confusion. Try it now: https://github.com/rxseger/rx_tools/commit/bbe75dea1346adca9a7b6f2cf74a8d9a180763d1

jonschoning, 2017-03-13:
that seems to have fixed `rx_fm` , thanks!

closed 2017-03-13 by jonschoning