rxseger/rx_tools #51: SDRPlay RSP2 not supported

dslotter, 2017-08-11:
The SDRPlay RSP1 and RSP2 are supported by SoapySDR, but do not apparently function with rx_fm.

```
dave@Moonquake:~/projects/hamradio/sdr/rx_tools$ rx_fm -v -f 97.1M
verbosity set to 1
SoapySDRDevice_make failed
Failed to open rtlsdr device matching (null).
```

SDRPlay has a GitHub repo here: [https://github.com/SDRplay](https://github.com/SDRplay)

guruofquality, 2017-08-11:
Can you make sure that [SoapySDRPlay](https://github.com/pothosware/SoapySDRPlay/wiki) is installed and that `SoapySDRUtil --find` sees the device. There may be an issue, but the snippet above just makes it look like the device isnt being found.

dslotter, 2017-08-11:
Here you go:

```
dave@Moonquake:~$ SoapySDRUtil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=3010 rev=0207 serno=<REDACTED> bus=002 port=003 devAvail=1
Found device 0
  driver = sdrplay
  label = SDRplay Dev0 RSP2 1702038E00

dave@Moonquake:~$ SoapySDRUtil --make
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Make device 
fwDownload: FW image size = 6083
mir_sdr_usb_GetDevices Dev0:vid=1df7 pid=3010 rev=0207 serno=<REDACTED> bus=002 port=003 devAvail=1
mir_sdr_usb_SetDeviceIdx idx=0 numDevices=1
Opened device with idVendor = 0x1df7 idProduct = 0x3010 fwVersion = 0x0207 busNum = 002 portNum = 003
mir_sdr_Init: starting hardware initialization
mir_sdr_Init: gR=60dB fs=2.000MHz rf=98.800MHz bw=1.536MHz if=0.000MHz
DownConvert: Enable=0 DecM=1 OutScale=0 (fs=2.000000 bw=1536 if=0)
mir_sdr_usb_USB DLL: Revision 0.1.1
mir_sdr_2500_Init: fnaddr = 2 detected, trying to change...
mir_sdr_2500_Init: fnaddr = 6
mir_sdr_2500_Init: adjusting squelch trim 0x1, rx gating enable 1, tx_trim 0, reg2 = 0x4801
initHw: Register7 = 0x000014
initRfFreqDependentHw(1): Tuner Register0 = 0x04f420
mir_sdr_GpioExtWrite: Addr=0 Data=0x00
mir_sdr_GpioExtWrite: Addr=1 Data=0x00
mir_sdr_GpioExtWrite: Addr=18 Data=0x5d
mir_sdr_GpioExtWrite: Addr=19 Data=0xa5
initRfFreqDependentHw(1): ExtGpioA = 0x5d ExtGpioB = 0xa5
mir_sdr_SetFs: Sample Freq requested 2000000.000000
mir_sdr_SetFs: Fs->FsNomHz+dFsHz=2000000.0+0.0Hz=2000000.0Hz FsToggle->1
mir_sdr_SetRf: f->98800000.000Hz (int=20 frac=af0 afc=0) fSynth:3161600000.000
mir_sdr_SetRf: Rf->RfNomHz+dRfHz+LifHz+Lif1Hz=98800000.0+0.0+0.0Hz+0.0Hz=98800000.0Hz RfToggle->1
mir_sdr_SetGr: GR->60[36,24,0,0] gRset->0x224 DCCALmode=3 DCCALspd=0 GrToggle->1
setToggleStates: initialising sampNum=0x00017a02, gainSetting=0x224, FsToggle=0, RfToggle=1, GrToggle=1
setToggleStates: initialising Fs=2000000.000, Rf=98800000.000, Gr=60
mir_sdr_2500_Readback001Trim: readback = 0xe402c0e
mir_sdr_2500_Bypass001Trim: reg13=0x1cd reg14=0x7002ce
  driver=SDRplay
  hardware=2 1702038E00
  mir_sdr_api_version=2.100000
  mir_sdr_hw_version=2
mir_sdr_usb_ReleaseDeviceIdx 0

```

dslotter, 2017-08-13:
While debugging the code, I found a workaround for SDRPlay RSP2. It appears that if I supply the command line argument "-d 0", then SoapySDRDevice_make succeeds, and the software functions.

guruofquality, 2017-08-13:
@dslotter You shouldn't need to pass in anything special when there is only one device, so ideally we fix that. Any details from your debugging?

FWIW, I noticed that we were passing a NULL pointer into the make call when no device args were specified, and so I fixed that: https://github.com/rxseger/rx_tools/commit/571cc9aeccb2004d81032cea484324aacb1a5852

Not sure if the NULL pointer was the cause of your problem, but passing in any device arguments (-d) would make it non-null and possibly fix the problem.

dslotter, 2017-08-13:
@guruofquality Changing the NULL pointer to a pointer to an empty string corrects the issue I originally reported. I believe it's safe to close this issue now. Thank you.

closed 2017-08-13 by dslotter