rxseger/rx_tools #52: rx_tools will not compile on kubuntu 17.10, maybe others.

rwgast, 2017-08-14:
After cloning rx_tools and making sure Pothos/SoapySDR is installed via the directions listed on there web page, I get this error when running "cmake ."

CMake Warning at CMakeLists.txt:16 (find_package):
  Could not find a package configuration file provided by "SoapySDR"
  (requested version 0.5) with any of the following names:

    SoapySDRConfig.cmake
    soapysdr-config.cmake

  Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
  "SoapySDR_DIR" to a directory containing one of the above files.  If
  "SoapySDR" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at CMakeLists.txt:18 (message):
  Soapy SDR development files not found...

-- Configuring incomplete, errors occurred!
See also "/home/robert/git-clones/rx_tools/CMakeFiles/CMakeOutput.log".

I am running PothosGUI, and soapy_power just fine using multiple SDR's. I also tried to follow the directions listed in the error above,

"  Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
  "SoapySDR_DIR" to a directory containing one of the above files.  If
  "SoapySDR" provides a separate development package or SDK, be sure it has
  been installed."

I had no idea where SoapySDRConfig.cmake or soapysdr-config.cmake would be so I just ran

sudo find / -type f -name "SoapySDRConfig.cmake" 
sudo find / -type f -name "soapysdr-config.cmake"

both find commands came up empty, these files do not exist on my computer, I have pothos/soapy installed along with almost every other SDR library and the dev packages.

Maybe some file names or dependencies have changed since RX_Tools was written? Im not sure but these programs will really help me get other SDR working with RTL only programs, Im hoping you can help.

Thanks in advance
RwGast

guruofquality, 2017-08-14:
If you installed soapysdr from apt-get, then you will also need development files: 
```
sudo apt-get install libsoapysdr-dev
```

But if you installed soapysdr from source, I would be curious to see the output of the make install command.

rwgast, 2017-08-14:
That was exactly the problem, I had assumed since pothosGUI worked I must have the dev tools too, guess not. 
 
The rx_tools build guide says install soapySDR and links to the soapy main page where you then follow download or build instructions. None of them seem to mention "libsoapysdr-dev", someone may think about adding the installation of that package to the rx_tools build page.

Thank you for the help works great now

guruofquality, 2017-08-15:
@rwgast I can try to update the wiki page/readme with better information. Where were you looking? rx_tools readme or a particular wiki url?

mkarwat, 2018-03-23:
I used:
https://github.com/rxseger/rx_tools
and then:
https://github.com/pothosware/PothosCore/wiki/Ubuntu
and had that problem, after running:
`sudo apt-get install libsoapysdr-dev`
everything was fine

guruofquality, 2018-03-24:
Thanks everyone. This is a pretty old bug time to close it out. The resolution is basically standard debian packaging practices. If you are using library foo from the package manager, and you want to develop against that library, you install libfoo-dev

closed 2018-03-24 by guruofquality