rxseger/rx_tools #53: Generic way to set gains

ha7ilm, 2017-09-08:
See: https://github.com/simonyiszk/openwebrx/issues/94

guruofquality, 2017-09-08:
I sure that nobody want to parse strings, but if we made the --gain option take a string, we could use the same markup used for device arguments to get a dictionary of gains. We can use an existing call `SOAPY_SDR_API SoapySDRKwargs SoapySDRKwargs_fromString(const char *markup);` to parse something like `-g "AMPL=12, VGA=4"` (for example). Its easy to iterate through a SoapySDRKwargs in C and apply each gain in the dictionary.

For the automatic/overall gain use case, --gain=16 will result in a SoapySDRKwargs with one entry with a key set to that string and an empty value -- so in this case we could know to use the setGain() with no named elements.

guruofquality, 2019-03-09:
this is the -g option in rx_fm (for example) it parses a gain string for key/value pairs of gain settings

closed 2019-03-09 by guruofquality