rxseger/rx_tools #54: Using Airspy: How can i activate bias-t?

runahue, 2017-09-15:
Hello friends:
Using Airspy.
How can i activate BIAS-T?
My airspy is running fine with rx_tools and Soapyairspy module, but i don't know how to activate it.
Thank's a lot.

john-, 2019-03-05:
I don't have any practical experience with this but take a look at airspy_gpio.   It has a reference to BiasT.   I don't know how to do this in rx_tools.

guruofquality, 2019-03-05:
So bias t is exposed in soapyairspay through writeSetting("biastee", "true"/"false")

My suggestion is to update this PR which allows for arbitrary settings through the command line: https://github.com/rxseger/rx_tools/pull/65

I dont know why it didnt get merged, but if you can make a new one I will merge it. The only suggestion, verbose_settings() should be simpler, we already have a function to parse comma/equals separate strings in SoapySDR, so sans the string parsing :-)

guruofquality, 2019-03-09:
Merged: https://github.com/rxseger/rx_tools/commit/0af9527a19582cfd805c943ecb60cc3f0f6c1be8

You can now do arbitrary settings in rx_sdr with the -t option. I suppose this would be good to add to the other executables, so please make a PR if needed :-)

closed 2019-03-09 by guruofquality