rxseger/rx_tools #56: Also.... Using rx_fm... When i play wav they are very slow.

runahue, 2017-12-31:
Doing more tests: When i play a wav file saved with rx_fm sound is very low using:
why?

pi@Raspi-Remote:~ $ rx_fm -M fm -f 94.8M -s 125000 -g LNA=10,MIX=10,VGA=10 -E deemp -F 9 -w 10000 -E wav kkk1.wav
Using device Airspy: device_id=0 origin=https://github.com/pothosware/SoapyAirspy
Found 1 antenna(s): RX
Found 3 gain(s): LNA MIX VGA
Found 1 frequencies: RF
Found 2 sample rates: 10000000 2500000
Found 0 bandwidths:
[INFO] Using format CS16.
Setting gain element LNA: 10.000000 dB
Setting gain element MIX: 10.000000 dB
Setting gain element VGA: 10.000000 dB
Set bandwidth parameter 10000 Hz.
WARNING: Failed to set offset tuning.
Tuned to 94800000 Hz.
Oversampling input by: 16x.
Oversampling output by: 1x.
Buffer size: 4.10ms
Sampling at 2000000 S/s.
Output at 125000 Hz.

c0debreaker, 2018-01-19:
Encountered the same. WOW, it's been 19 days and no replies yet

guruofquality, 2018-01-19:
I saw this: https://github.com/rxseger/rx_tools/issues/37

> Thanks, using SoapyAirspy the device is now used but i have a few other problems the audio plays very slowly.

> rx_fm -f 99.3M -M fm -s 170k -A fast -r 32k -l 0 -E deemp -w 2500000 -g LNA=11,MIX=5,IF=8| play -r 300k -t raw -e s -b 16 -c 1 -V1 -

> using play -r 300k it's playing at the right kind of speed.
if i don't use -w 2500000 is plays the wrong station 97.4MHz rather than 99.3MHz, i think the offset tuning is a bit off.

So maybe its the output rate `-r` ? Just a guess, but probably someone with an airspy needs help debug this to narrow down the issue.

c0debreaker, 2018-01-19:
I have an airspy and I can try different settings. I tried different -r values last night with no luck.

c0debreaker, 2018-02-09:
I still have no luck! Anyone had luck with this tool?