rxseger/rx_tools #58: OError:  reading stream -4

Kishaihd, 2018-03-24:
Running `rx_power -f 88M:108M:200k -e 10m -i1 rx_power_test.csv` on the Raspberry Pi 3
works 4 out of 5 times, but occasionally, I just get a constant stream to the terminal alternating between 
`OError: reading stream -4` and `Error: reading stream -1`. 
I've searched the documentation in libhackrf, the Hackrf One github page, and this repo, but I'm unable to find anything explaining what this error is. 

Any ideas?

zodoczi, 2019-11-14:
I got similar errors, any idea guys? Thanks

```
rx_power.exe -f 10M:100M:2M -d driver=hackrf -i 10 -1

...

Error: bad retune at 11000000 Hz (4 of 3 attempts), r=-5, flags=0 (try increasing -S or -R).
Error: reading stream -5
Error: bad retune at 13000000 Hz (4 of 3 attempts), r=-5, flags=0 (try increasing -S or -R).
Error: reading stream -5
Error: bad retune at 15000000 Hz (4 of 3 attempts), r=-5, flags=0 (try increasing -S or -R).
Error: reading stream -5
```

zodoczi, 2019-11-14:
forgot to mention that using Windows 10 (and PothosSDR as driver base)

zodoczi, 2019-11-14:
rx_fm still gives the same error with hackrf:
`C:\Program Files\PothosSDR\bin>rx_fm.exe -f 90.9M -d driver=hackrf
dup2 start: No error
[INFO] Opening HackRF One #0 308066e622c0464b...
Using device HackRF One: clock source=internal part id=a000cb3c005c475c serial=0000000000000000308066e622c0464b version=2018.01.1
Found 1 channel(s) :
Channel 0 :
  Found 1 antenna(s): TX/RX
  Found 3 gain(s): LNA AMP VGA
  Found 1 frequencies: RF
  Found 20 sample rates: 1000000 2000000 3000000 4000000 5000000 6000000 7000000 8000000 9000000 10000000 11000000 12000000 13000000 14000000 15000000 16000000 17000000 18000000 19000000 20000000
  Found 16 bandwidths: 1750000 2500000 3500000 5000000 5500000 6000000 7000000 8000000 9000000 10000000 12000000 14000000 15000000 20000000 24000000 28000000
Bandwidth set to automatic resulted in 0 Hz.
Tuned to 91152000 Hz.
Oversampling input by: 42x.
Oversampling output by: 1x.
Buffer size: 8.13ms
Sampling at 1008000 S/s.
Output at 24000 Hz.
dup2 stop: No error
readStream read failed: -5
dongle_thread_fn terminated
Signal caught, exiting!`