rxseger/rx_tools #60: rx_sdr won't work with AirspyHF+

Gymnae, 2018-09-26:
I compiled the latest versions of all required packages, but rx_sdr won't work:

SoapySDR and SoapyRemote work, I'm using it with gqrx:
 ```
SoapySDRUtil --probe="driver=remote,type=airspyhf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=remote,type=airspyhf

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=AirspyHF
  hardware=AirspyHF
  device_id=0
  origin=https://github.com/sevo/SoapyAirspyHF
  remote:version=0.5.0-g02e29496

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CF32
  Native format: CF32 [full-scale=3.4e+38]
  Stream args:
     * Remote Format - The stream format used on the remote device.
       [key=remote:format, default=CF32, type=string, options=(CF32)]
     * Remote Scale - The factor used to scale remote samples to full-scale floats.
       [key=remote:scale, default=339999999999999996123846586046231871488.000000, type=float]
     * Remote MTU - The maximum datagram transfer size in bytes.
       [key=remote:mtu, units=bytes, default=1500, type=int]
     * Remote Window - The size of the kernel socket buffer in bytes.
       [key=remote:window, units=bytes, default=44040192, type=int]
     * Remote Priority - Specify the scheduling priority of the server forwarding threads.
       [key=remote:priority, default=0.500000, type=float, range=[-1, 1]]
     * Remote Protocol - Specify the transport protocol for the remote stream.
       [key=remote:prot, default=udp, type=string, options=(udp, tcp, none)]
  Antennas: RX
  Full gain range: [0, 0] dB
  Full freq range: [0.009, 31], [60, 260] MHz
    RF freq range: [0.009, 31], [60, 260] MHz
  Sample rates: 0.768 MSps
```
But running rx_sdr I get the following. In this example I defined a long run command, but even just `remote=localhost` yields the very same result:

```
rx_sdr -g 0 -f 350000 -F CF32 -s 768000 -d driver=remote,remote:driver=airspyhf,remote=localhost,remote:format=CF32,remote:scale=339999999999999996123846586046231871488.000000  - > test.iq
Using device AirspyHF: device_id=0 origin=https://github.com/sevo/SoapyAirspyHF remote:version=0.5.0-g02e29496 
Found 1 antenna(s): RX 
Found 0 gain(s): 
Found 1 frequencies: RF 
Found 1 sample rates: 768000 
Found 0 bandwidths: 
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CS16, scaleFactor=32768, mtu=1500, window=44040192)
[INFO] Client side stream bound to 127.0.0.1:60058
[INFO] Client side status bound to 127.0.0.1:43682
SoapySDRDevice_setupStream failed
Failed to open sdr device matching 'driver=remote,remote:driver=airspyhf,remote=localhost,remote:format=CF32,remote:scale=339999999999999996123846586046231871488.000000'

```

guruofquality, 2018-10-18:
* Related: https://github.com/pothosware/SoapyAirspyHF/issues/1

bshaffer82, 2020-11-12:
Airspy HF+ will work via Soapy now if all are built from source.  
Rx_power does not actually work properly.  

I don't fully understand why it "almost works" if I leave it run for a few hours I do see something that kind of looks like spectra but I think the gain is wrong or the bin size isn't right.

These are my concern area's:

 rx_power -f 3.6M:4.0M:1000
Number of frequency hops: 1           
Dongle bandwidth: 2800000Hz                 <--I don't think 2.8 Ghz is correct for this dongle
Downsampling by: 7x                                 <--That kind of implies this downsample might be incorrect (not sure)
Cropping by: 0.00%
Total FFT bins: 512
Logged FFT bins: 512                                    
FFT bin size: 781.25Hz
Buffer size: 16384 bytes (2.93ms)
Reporting every 10 seconds
Using device AirspyHF: serial=c852eb5defcb39cc
Found 1 channel(s) :
Channel 0 :
  Found 1 antenna(s): RX
  Found 2 gain(s): LNA RF                          <-I have tried -g (50, RF=50, LNA=50) but none move my signal above -18
  Found 1 frequencies: RF
  Found 6 sample rates: 912000 768000 456000 384000 256000 192000   <--which one is rx_power choosing? does it matter
  Found 0 bandwidths:                                <- Zero bandwidths??0
2020-11-12, 16:26:49, 3600000, 4000000, 781.25, 62937, -18.66, -18.55, -18.66, -18.60, -18.68, -18.62, -18.65, -18.68, -18.54, -18.57, -18.54, -18.59, -18.66, -18.74, -18.59, -18.62, -18.61, -18.78, -18.95, -18.74, -18.70, -18.76, -18.69, -18.55, -18.68, -18.75, -18.64, -18.64, -18.82, -18.56, -18.59, -18.64, -18.58, -18.53, -18.59, -18.58, -18.65, -18.74, -18.60, -18.48, -18.69, -18.75, -18.69, -18.71, -18.69,

As you can see I'm getting some very small variation in the signals but not enough.  Using SDR# I've got a very dynamic 80m environment with lots of noise, spikes, strong and weak signals

This is all I get out after running the rx_power output through heatmap.py
[image]

What I wanted to do was use this tool to hunt down noise sources which I see readily in my SDR-console waterfall after 11pm but I just didn't want to stay up all night to determine when they stop.