rxseger/rx_tools #63: No sound in output

rrobinett, 2018-11-09:
I am trying to replace rtl-fm with rx_fm on a Raspberry Pi 3B in order to benefit from improved tuning  frequency accuracy and access to all Soapy-supported devices.

Running 'rtl_fm  -M wbfm -f 88.5M | sox -r 32k -t raw -e s -b 16 -c 1 - -d' I can clearly hear my FM station.

However, running 'rx_fm  -M wbfm -f 88.5M | sox -r 32k -t raw -e s -b 16 -c 1 - -d' I hear dead silence.  There are no errors reported and there is clearly an rx_fm output bitstream, so I seem to have misconfigured rx_fm.:

------------------
pi@PiB83:/tmp/wspr-captures/AI6VN_VHF_0/1 $ rx_fm  -M wbfm -f 88.5M | sox -r 32k -t raw -e s -b 16 -c 1 - -d

-: (raw)

  Encoding: Signed PCM
  Channels: 1 @ 16-bit
Samplerate: 32000Hz
Replaygain: off
  Duration: unknown

In:0.00% 00:00:00.00 [00:00:00.00] Out:0     [      |      ]        Clip:0    Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
Using device R820T: origin=https://github.com/pothosware/SoapyRTLSDR rtl=0
Found 1 antenna(s): RX
Found 1 gain(s): TUNER
Found 2 frequencies: RF CORR
Found 10 sample rates: 250000 1024000 1536000 1792000 1920000 2048000 2160000 2560000 2880000 3200000
Found 0 bandwidths:
[INFO] Using format CS16.
Tuner gain semi-automatically set to 40 dB
Bandwidth set to automatic resulted in 0 Hz.
Tuned to 88771000 Hz.
Oversampling input by: 6x.
Oversampling output by: 1x.
Buffer size: 8.03ms
Exact sample rate is: 1020000.026345 Hz
Sampling at 1020000 S/s.
Output at 170000 Hz.
In:0.00% 00:00:00.51 [00:00:00.00] Out:8.19k [      |      ]        Clip:0    sox WARN alsa: under-run
In:0.00% 00:00:00.77 [00:00:00.00] Out:16.4k [      |      ]        Clip:0    sox WARN alsa: under-run
In:0.00% 00:00:13.57 [00:00:00.00] Out:426k  [      |      ]        Clip:0    sox WARN alsa: under-run
In:0.00% 00:00:23.81 [00:00:00.00] Out:754k  [      |      ]        Clip:0    ^CSignal caught, exiting!

User cancel, exiting...
readStream read failed: -1
dongle_thread_fn terminated
In:0.00% 00:00:24.03 [00:00:00.00] Out:762k  [      |      ]        Clip:0
Aborted.
pi@PiB83:/tmp/wspr-captures/AI6VN_VHF_0/1 $

rrobinett, 2019-02-16:
I continue to have this problem

vdleun, 2019-08-12:
I had the same problem with my sdrplay rsp1a. I solved it by including the tuner bandwidth (-w 200000) and sdr sample rate (-s 200k). 

rx_fm -M nfm -f 144.800M -d driver=sdrplay -w 200k -s 500k -r 32k -g 50 | direwolf -c sdr-1200bps.conf -r 32000 -D 1 -B 1200 -

By firing up rx_fm this way, I now have audio and am piping it into direwolf to decode APRS packets. I checked the audio by running the samen rx_fm command and piping it into sox. Before -w and -s it was dead silence. The values of -w and -s parameters are taken from the output of rx_fm.

  Found 1 antenna(s): RX
  Found 2 gain(s): IFGR RFGR
  Found 2 frequencies: RF CORR
  Found 13 sample rates: 250000 500000 1000000 2000000 2048000 3000000 4000000 5000000 6000000 7000000 8000000 9000000 10000000
  Found 8 bandwidths: 200000 300000 600000 1536000 5000000 6000000 7000000 8000000