rxseger/rx_tools #68: Failed to activate stream

bclswl0827, 2019-02-06:
My decvice is SDRPlay RSP1. I installed its driver and packages step by step.

But when I run the command `rx_sdr - | csdr through > /dev/null` to test if everything is all right, I got the error

```
root@yukiho:~# rx_sdr - | csdr through > /dev/null
Using device B0002P0001: mir_sdr_api_version=2.130000 mir_sdr_hw_version=1
Found 1 antenna(s): RX
Found 2 gain(s): IFGR RFGR
Found 2 frequencies: RF CORR
Found 13 sample rates: 250000 500000 1000000 2000000 2048000 3000000 4000000 5000000 6000000 7000000 8000000 9000000 10000000
Found 8 bandwidths: 200000 300000 600000 1536000 5000000 6000000 7000000 8000000
[INFO] Using format CS16.
Using output format: CU8 (input format CS16)
Sampling at 2048000 S/s.
Tuned to 100000000 Hz.
Reading samples in sync mode...
Failed to activate stream
```

I blacklisted the kernel modules correctly, but it still...

```
root@yukiho:~# SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=B0002P0001
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=1, type=string, options=(0, 1, 2, 3)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 42] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 3] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```
It seems the driver is correctly installed, so I think there's something problem with rx_tools, or some dependences is not installed in my Pi...

There's no such problem like this, so I got puzzled... Help...

guruofquality, 2019-02-08:
You may want to report this to https://github.com/pothosware/SoapySDRPlay/issues 
I have a SDRPlay and this is working on my x86 machine, so it could be PI related.

bclswl0827, 2019-02-08:
Got it, thanks.

closed 2019-02-08 by bclswl0827