rxseger/rx_tools #70: latest commit break hackrf

Anderl94, 2019-02-21:
With the latest commit my HackRF failt with rx_fm.

Log:
rx_fm -f 144M -s 48000 -v 3
verbosity set to 1
[INFO] Opening HackRF One #0 308066e62355214b...
Using device HackRF One: clock source=internal part id=a000cb3c0045434f serial=0000000000000000308066e62355214b version=2017.02.1 
Found 1 channel(s) :
Channel 0 :
  Found 1 antenna(s): TX/RX 
  Found 3 gain(s): LNA AMP VGA 
  Found 1 frequencies: RF 
  Found 20 sample rates: 1000000 2000000 3000000 4000000 5000000 6000000 7000000 8000000 9000000 10000000 11000000 12000000 13000000 14000000 15000000 16000000 17000000 18000000 19000000 20000000 
  Found 16 bandwidths: 1750000 2500000 3500000 5000000 5500000 6000000 7000000 8000000 9000000 10000000 12000000 14000000 15000000 20000000 24000000 28000000 
Invalid channel selected
Bandwidth set to automatic resulted in 0 Hz.
downsample_passes = 0 (= # of fifth_order() iterations), downsample = 21
capture_rate = dm->downsample * dm->rate_in = 21 * 48000 = 1008000
optimal_settings(freq = 144000000): capture_freq = freq + capture_rate/4 = 144252000
optimal_settings(freq = 144000000): capture_freq +=  cs->edge * dm->rate_in / 2 = 0 * 48000 / 2 = 144252000
optimal_settings(freq = 144000000) delivers freq 144252000, rate 1008000
verbose_set_frequency(144252000 Hz)
  frequency is away from parametrized one, to avoid negative impact from dc
Tuned to 144252000 Hz.
Oversampling input by: 21x.
Oversampling output by: 1x.
Buffer size: 8.13ms
verbose_set_sample_rate(1008000 Hz)
Sampling at 1008000 S/s.
Output at 48000 Hz.
readStream read failed: -5
dongle_thread_fn terminated

john-, 2019-02-24:
I just recently started trying to use rx_tools and have not been able to get it to work due to same error.   It happens with airspy as well as hackrf.

guruofquality, 2019-02-24:
@fmoessbauer I have no idea how this compiled, the call to verbose_setup_stream was completely wrong. I pushed a fix, can you guys also try it (tested fm on my bladerf2)

john-, 2019-02-24:
That fixed it for me (tested on hackrf).

fmoessbauer, 2019-02-24:
@guruofquality: I'm really sorry for breaking rx_tools. When squashing the commits to prepare the PR, I lost the changes in the staging area... And unfortunately the broken state does compile (at leats on gcc 7.3.0).

guruofquality, 2019-02-24:
No worries, rx tools just needed some TLC

closed 2019-02-24 by guruofquality