rxseger/rx_tools #79: rtl_sdr.c: Failed to activate stream, WARNING: sync read failed

metssigadus, 2019-07-31:
Crossposting an issue b/c it happens when various packages interact. The problem is described here: https://github.com/pothosware/SoapySDR/issues/232 .

**Symptoms**: In a new installation:

-  "SoapySDRUtil --find" works properly, all the SDRs (1 RTL-SDR, 2x SDRPlay) with their attributes are recognised (including remotely);
-   "rtl_test" and rtl_fm work correctly. No errors with an RTL-SDR stick and rtl_tools;
-   With Soapy tools, the command part works well (SDRs are visible, attributes are OK), but the streaming does never start. Errors are: "WARNING: sync read failed. -1" ; "Failed to activate stream". SoapyRemote server is working and able to transport the "sync read failed" errors to a remote Soapy host, and OpenWebRx (or to be more precise - rx_sdr) there reprints these.

I apologise if this issue belongs to elsewhere.

bclswl0827, 2019-07-31:
https://github.com/rxseger/rx_tools/issues/68
Having same issue on my SDRPlay RSP1, I use Orange Pi Lite. But soon I found out that it might be a problem with my device performance being too low, because it runs well on x86…
Hope this can help you.

metssigadus, 2019-07-31:
Thnx for assistance. But the reasons likely are elsewhere. The host is an " LENOVO 96417KG/LENOVO, BIOS 2QKT15AUS 01/05/2007" with an "Intel(R) Pentium(R) D CPU 3.00GHz", Total of 2 processors (11970.22 BogoMIPS). Should be powerful enough.
Tried with _strace_ - resulting nothing I am capable to dig into. 

The log from the  _**strace rx_sdr -d 0 -**_:
setsockopt(9, SOL_SOCKET, SO_RCVBUF, [44040192], 4) = 0
getsockopt(9, SOL_SOCKET, SO_RCVBUF, [88080384], [4]) = 0
write(2, "[INFO] Configured receiver endpo"..., 96[INFO] Configured receiver endpoint: dgram=1452 bytes, 714 elements @ 2 bytes, window=43008 KiB
) = 96
sendto(9, "\0\0\0\30\0\0\0\0\0\0r\260\0\0\0\0\0\0\0\0\0\0\0\0", 24, MSG_NOSIGNAL, NULL, 0) = 24
write(2, "Reading samples in sync mode...\n", 32Reading samples in sync mode...
) = 32
sendto(7, "SRPC\0\0\4\0\0\0\0.\17\2\0\0\1.\2\0\0\0\0\2\0\0\0\0\3\0\0\0"..., 46, MSG_NOSIGNAL, NULL, 0) = 46
select(8, [7], NULL, NULL, {tv_sec=3, tv_usec=0}) = 1 (in [7], left {tv_sec=2, tv_usec=989442})
recvfrom(7, "SRPC\0\0\4\0\0\0\0\25", 12, MSG_WAITALL, NULL, NULL) = 12
recvfrom(7, "\2\0\0\0\0CPRS", 9, 0, NULL, NULL) = 9
dup2(3, 1)                              = 1
sendto(9, "\0\0\0\30\0\0\0\0\0\0r\260\0\0\0\0\0\0\0\0\0\0\0\0", 24, MSG_NOSIGNAL, NULL, 0) = 24
select(10, [9], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
write(2, "WARNING: sync read failed. -1\n", 30WARNING: sync read failed. -1
) = 30
sendto(9, "\0\0\0\30\0\0\0\0\0\0r\260\0\0\0\0\0\0\0\0\0\0\0\0", 24, MSG_NOSIGNAL, NULL, 0) = 24
select(10, [9], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
write(2, "WARNING: sync read failed. -1\n", 30WARNING: sync read failed. -1
) = 30
sendto(9, "\0\0\0\30\0\0\0\0\0\0r\260\0\0\0\0\0\0\0\0\0\0\0\0", 24, MSG_NOSIGNAL, NULL, 0) = 24
select(10, [9], NULL, NULL, {tv_sec=1, tv_usec=0}) = 0 (Timeout)
write(2, "WARNING: sync read failed. -1\n", 30WARNING: sync read failed. -1
) = 30
[..]

bclswl0827, 2019-08-01:
Do you use USB Hub to connect them?
…try to connect directly?

metssigadus, 2019-08-01:
No USB hubs currently in use. SDRs directly at the host. BTW I have a positive experience with SDRPlays plus two 20m extenders to the roof (altogether 42m of distance) plus a hub of chosen quality with elevated (5.2V) external VCC. 10MHz bandwidth was not a problem - sustainability was tested. However, this is maximum. Yet another hub will be denied by the drivers.
My current problem very likely is from the software world.