rxseger/rx_tools #83: SoapyRemote driver doesn't seem to work?

Aang23, 2020-01-16:
Using rx_fm with a remote device (hackrf) string doesn't seem to work.

I first tested with the following command, which worked flawlessly.
`rx_fm -d driver=hackrf -f 98.4M -M wbfm -s 200000 -r 24000 - | aplay -r 24000 -f S16_LE`

Then the same HackRF was plugged into a known-working SoapySDRServer (works with CubicSDR / Gqrx / GNURadio), and I ran the same command specifying the proper device :
`rx_fm -d "driver=remote, remote=192.168.0.23, remote:driver=hackrf" -f 98.4M -M wbfm -s 200000 -r 24000 - | aplay -r 24000 -f S16_LE`
It connected, I could see data flowing in through the network, but nothing playing, and I got the following warning
`[WARNING] Set thread priority 0.5 failed: Function not implemented`
Trying to kill the process by ^C caused it to hang out.
https://paste.altillimity.com/vopumibibu.coffeescript

guruofquality, 2020-01-17:
Since the hackrf is streaming complex int8 samples over the network you might just be seeing a different of the byte ending up in the bottom of the 16-bits vs the top depending upon what soapyremote does vs soapyhackrf when converting the number.

Its just a guess, but you can pass in the stream args or modify the samples in rx_fm, that might be a worthwhile test. It might be a small code change to do it: https://github.com/pothosware/SoapyRemote/wiki#remote-scale

Aang23, 2020-02-27:
Took a while but I tested this again passing stream arguments / changing some code and didn't manage to get anything working with the HackRF.
I tried it out with my RTLSDR dongle as well, which surprisingly didn't work in the exact same way. (While CubicSDR or GnuRadio works flawlessly)

RDPowerz, 2020-12-28:
Did some similar tests with rtlsdr dongle locally and airspyhf dongle remotely and could not find a case that worked with rx_fm, but changing over to using rtl_fm with same command line arguments as rx_fm and (of course) using the local rtlsdr dongle worked.

Did use rx_sdr to save IQ to file then used gnuradio file source plus frequency sink and waterfall sink to look at the IQ and it looked fine for both local rtlsdr and remote airspyhf so I suspect the problem lies within rx_fm...