rxseger/rx_tools #88: rx_fm problem with LimeSDR mini

ph03n1xAu, 2020-11-11:
Has anyone got any success using rx_fm script to work with LimeSDR mini.

I followed everything as per instructions, but I’m getting this error on my debian system. Though it is not relevant, just as a side note, I can run gnuradio scripts with limesdr mini without any problem in debian.

readStream read failed: -1
dongle_thread_fn terminated

snippet of error:

[image]

Appreciate any suggestion:-)

akaBaib, 2021-02-26:
Hi @ph03n1xAu, I have the same problem. Did you find a solution?

ph03n1xAu, 2021-03-10:
Hi @akaBaib, 

No, I didn't find any solution yet..in fact I was waiting with a hope that someone here will be able to guide me. I checked the source code for this error but didn't quite able to understand the underlying issue.