rxseger/rx_tools #90: rx_fm fails with local rtlsdr device whereas rtl_fm works using the same command line arguments

RDPowerz, 2020-12-28:
Am now finding that I found a way to get rtl_fm (but not rx_fm) to decode the local classical fm station (using rtl-sdr dongle of course).  Still can't get rx_fm (i.e. soapy version) to work.  They are using the same code to parse the command lines, so I can change 'rtl_fm' to 'rx_fm' and it should just work, but it does not.   It is my understanding that they are supposed to be command line compatible.

Works:
rtl_fm -M wbfm -f 99.5M | play -r 32k -t raw -e s -b 16 -c 1 -V1 -

Doesn't work;
rx_fm -M wbfm -f 99.5M | play -r 32k -t raw -e s -b 16 -c 1 -V1 -

Others should be able to try these two cases just by changing the -f argument to tune to a strong local station.

Version info:

rx_tools$ git log | head -n1
commit 811b21c4c8a592515279bd19f7460c6e4ff0551c

Soapy info:

rx_tools$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g926c86d9
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so   (0.3.3-7d53087)
Module found: /usr/local/lib/SoapySDR/modules0.8/libairspyhfSupport.so (0.1.1-b6cfbf5)
Module found: /usr/local/lib/SoapySDR/modules0.8/libaudioSupport.so    (0.1.1-9aa2b5c)
Module found: /usr/local/lib/SoapySDR/modules0.8/libremoteSupport.so   (0.6.0-c09b2f1)
Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so   (0.3.1-24b27fa)
Available factories... airspyhf, audio, hackrf, remote, rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

Working use case:
rx_tools$ rtl_fm -M wbfm -f 99.5M | play -r 32k -t raw -e s -b 16 -c 1 -V1 -
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM

-: (raw)

 File Size: 0         
  Encoding: Signed PCM    
  Channels: 1 @ 16-bit   
Samplerate: 32000Hz      
Replaygain: off         
  Duration: unknown      

In:0.00% 00:00:00.00 [00:00:00.00] Out:0     [      |      ]        Clip:0    Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Tuned to 99771000 Hz.
Oversampling input by: 6x.
Oversampling output by: 1x.
Buffer size: 8.03ms
Exact sample rate is: 1020000.026345 Hz
Sampling at 1020000 S/s.
Output at 170000 Hz.
In:0.00% 00:00:02.30 [00:00:00.00] Out:65.5k [  -===|===-  ] Hd:5.6 Clip:0    

Failing use case: 

rxtools$ rx_fm -M wbfm -f 99.5M | play -r 32k -t raw -e s -b 16 -c 1 -V1 -

-: (raw)

 File Size: 0         
  Encoding: Signed PCM    
  Channels: 1 @ 16-bit   
Samplerate: 32000Hz      
Replaygain: off         
  Duration: unknown      

In:0.00% 00:00:00.00 [00:00:00.00] Out:0     [      |      ]        Clip:0    
RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

Found Rafael Micro R820T tuner
Using device Audio: device_id=13 origin=https://github.com/pothosware/SoapyAudio 
Found 1 channel(s) :
Channel 0 :
  Found 1 antenna(s): RX 
  Found 0 gain(s): 
  Found 1 frequencies: RF 
  Found 14 sample rates: 4000 5512 8000 9600 11025 16000 22050 32000 44100 48000 88200 96000 176400 192000 
  Found 0 bandwidths: 
[INFO] Using format CS16.
Bandwidth set to automatic resulted in 0 Hz.
Tuned to 99771000 Hz.
Oversampling input by: 6x.
Oversampling output by: 1x.
Buffer size: 8.03ms
Sampling at 1020000 S/s.
Output at 170000 Hz.

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

RtApiAlsa: dump hardware params just after device open:

ACCESS:  RW_INTERLEAVED
FORMAT:  U8 S16_LE S16_BE S24_LE S24_BE S32_LE S32_BE FLOAT_LE FLOAT_BE MU_LAW A_LAW S24_3LE S24_3BE
SUBFORMAT:  STD
SAMPLE_BITS: [8 32]
FRAME_BITS: [8 1024]
CHANNELS: [1 32]
RATE: [1 192000]
PERIOD_TIME: (5 4294967295)
PERIOD_SIZE: [1 1398102)
PERIOD_BYTES: [128 1398102)
PERIODS: [3 1024]
BUFFER_TIME: (15 4294967295]
BUFFER_SIZE: [3 4194304]
BUFFER_BYTES: [384 4194304]
TICK_TIME: ALL

RtApiAlsa: dump hardware params after installation:

ACCESS:  RW_INTERLEAVED
FORMAT:  FLOAT_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 32
CHANNELS: 1
RATE: 192000
PERIOD_TIME: (10666 10667)
PERIOD_SIZE: 2048
PERIOD_BYTES: 8192
PERIODS: 4
BUFFER_TIME: (42666 42667)
BUFFER_SIZE: 8192
BUFFER_BYTES: 32768
TICK_TIME: [0 0]

RtApiAlsa: dump software params after installation:

tstamp_mode: NONE
tstamp_type: GETTIMEOFDAY
period_step: 1
avail_min: 2048
start_threshold: 2048
stop_threshold: -1
silence_threshold: 0
silence_size: 4611686018427387904
boundary: 4611686018427387904
In:0.00% 00:00:01.02 [00:00:00.00] Out:24.6k [ -====|====- ]        Clip:0

rrobinett, 2021-01-03:
Your two lines work perfectly and identically on my 4GB Raspberry Pi 4 running buster ;=(

RDPowerz, 2021-01-03:
Interesting!  
Can you provide some more info, such as output of 'SoapySDRUtil --info' and 'SoapySdrUtil --find' on your RPi?

This will tell me what software and hardware you are using.

Regards,RDP

rrobinett, 2021-01-03:
The information you requested is attached below.
I have also just learned that usb mode (Upper Sideband) requires a source code fix to remove from rx_sdr a divide by 256.
Perhaps that change would help FM modes too.

+++++++++++++++++++++++ 

pi@Wsprdaemon-Pi:~ $ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g926c86d9
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so (0.3.1-24b27fa)
Available factories... rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
pi@Wsprdaemon-Pi:~ $

pi@Wsprdaemon-Pi:~ $ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Found device 0
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  serial = 00000001
  tuner = Rafael Micro R820T

pi@Wsprdaemon-Pi:~ $

RDPowerz, 2021-01-03:
Very interesting!  

You have the same soapy library and soapy rtl driver versions I am using.

Also it seems rx_tools have not changed in a while so it seems safe to say we have the same version of these.

I am using Ubuntu 18.04 in x86_64 build mode.

rx_fm is not producing a version number so I went to my build directory for rx_tools and did:

rx_tools$ git log | head -n1
commit 811b21c

That commit maps to March 2019 so it seems we would have the same rx_tools as well.

I'm at a loss to explain the difference.

I plan to build a rpi based system soon, that should give me another data point.

Regards,
RDP