rxseger/rx_tools #103: Can somebody here just post a compiled Windows copy?

BenHut1, 2024-07-03:
I have Windows not Linux, so I don't have access to the compiling tools described here. Can someone here compile it for Windows (using the Linux compiler tools in cross-compiling mode to make Windows executables) for me? I have an AirSpy SDR dongle, and on Windows the official software (SDR Sharp) is kinda slow and drops buffers on the AirSpy's highest bandwidth (10MHz) due to all the extra processing it does to bring you live demodulated audio and spectrogram. So I was hoping for a commandline tool like rtl-sdr.exe or rtl-fm.exe (which are of course designed for an RTL SDR) but remade to work with an AirSpy. And I heard that rx_tools is just what I'm looking for. I need it to just dump raw complex samples to a file for later demodulation with other software. This way my AirSpy shouldn't drop any buffers. Unfortunately this rx_tools thing is only offered in source code. So that's why I'm asking for someone to post a compiled EXE file here for me to use.

eabase, 2025-02-25:
I don't know if it works for Airspy as I can only see support for RTL-SDR...and maybe HackRF (`-d` option).

But [here](https://github.com/eabase/TheLostOnes/blob/main/rx_tools_x64.7z) is fresh compile.

unixpunk, 2025-02-25:
rx_tools uses SoapySDR, so its compatible with almost every SDR as long as you have the modules loaded.